Juliankautsch/F19 #120: Design basic steering geometry

Maciejgrybko, 2018-10-17:
- reproduce steering geometry from this video: https://www.youtube.com/watch?v=sUI1gryNUqE
- use last year's parameters for: wheel size and front track

Recommended video: https://www.youtube.com/watch?v=Z1Y14AejfQU

DomVaitkus, 2018-10-18:
Hi, I said I'll do this. Can someone add me to zenhub

TianZH19, 2018-10-23:
[Basic Steering Geometry.zip](https://github.com/Juliankautsch/F19/files/2507546/Basic.Steering.Geometry.zip)
Here's the basic steering geometry including assembly parts in solidworks.
[Block.zip](https://github.com/Juliankautsch/F19/files/2507548/Block.zip)
[Parts.zip](https://github.com/Juliankautsch/F19/files/2507549/Parts.zip)

closed 2018-10-24 by Maciejgrybko