Juliankautsch/F19 #145: Cut Tubes

Juliankautsch, 2018-11-02:
Please comment below which tubes you have cut so we can track progress (copy the list from here and mark done after the tube name):
Tube 1 (Done)
Tube 2 (Done)
Tube 3
Tube 4
Tube 5
Tube 6
Tube 7
Tube 8
Tube 9
Tube 10
Tube 11
Tube 12
Tube 13
Tube 14
Tube 15
Tube 16
Tube 17
Tube 18
Tube 19
Tube 20
Tube 21
Tube 22
Tube 23
Tube 24
Tube 25
Tube 26
Tube 27
Tube 28
Tube 29
Tube 30
Tube 3
Tube 3
Tube 3
Tube 3
Tube 3
Tube 3
Tube 3
Tube 3

RyanMudaliar, 2018-11-08:
Please comment below which tubes you have cut so we can track progress (copy the list from here and mark done after the tube name):
Tube 1 (Done)
Tube 2 (Done)
Tube 3
Tube 4
Tube 5
Tube 6
Tube 7
Tube 8
Tube 9
Tube 10
Tube 11
Tube 12
Tube 13
Tube 14 (Done)
Tube 15
Tube 16
Tube 17
Tube 18
Tube 19
Tube 20 (Done)
Tube 21
Tube 22
Tube 23
Tube 24
Tube 25
Tube 26
Tube 27
Tube 28
Tube 29 (Done)
Tube 30 (Done)
Tube 31 (Done)
Tube 32 (Done)
Tube 33
Tube 34
Tube 35
Tube 36
Tube 37
Tube 38
Tube 39

samuelhodgkinson, 2018-11-08:
Please comment below which tubes you have cut so we can track progress (copy the list from here and mark done after the tube name):
Tube 1 (Done)
Tube 2 (Done)
Tube 3
Tube 4
Tube 5
Tube 6
Tube 7
Tube 8
Tube 9
Tube 10
Tube 11
Tube 12
Tube 13
Tube 14 (Done)
Tube 15
Tube 16
Tube 17
Tube 18
Tube 19
Tube 20 (Done) *2
Tube 21 
Tube 22
Tube 23
Tube 24
Tube 25
Tube 26
Tube 27
Tube 28 (Done) *2
Tube 29 (Done)
Tube 30 (Done) *2
Tube 31 (Done)
Tube 32 (Done)
Tube 33 (Done) *2
Tube 34 (Done) *2
Tube 35 (Done) *2
Tube 36
Tube 37
Tube 38
Tube 39

Please put the quantity of tubes you cut to prevent duplication. (As above)

FilipMilto, 2018-11-17:
Please comment below which tubes you have cut so we can track progress (copy the list from here and mark done after the tube name):
Tube 1 (Done)
Tube 2 (Done)
Tube 3 (Done)*2
Tube 4 (Done)*2
Tube 5
Tube 6 (Done)
Tube 7
Tube 8 (Done)*2
Tube 9 (Done)*2
Tube 10 (Done)
Tube 11
Tube 12
Tube 13
Tube 14 (Done)
Tube 15
Tube 16
Tube 17 (Done)
Tube 18 (Done)
Tube 19
Tube 20 (Done) *2
Tube 21 (Done)
Tube 22
Tube 23
Tube 24
Tube 25
Tube 26
Tube 27
Tube 28 (Done) *2
Tube 29 (Done)
Tube 30 (Done) *2
Tube 31 (Done)
Tube 32 (Done)
Tube 33 (Done) *2
Tube 34 (Done) *2
Tube 35 (Done) *2
Tube 36 (Done)*2
Tube 37 (Done)*2
Tube 38
Tube 39

Please put the quantity of tubes you cut to prevent duplication. (As above)

closed 2018-11-18 by Juliankautsch